tio/tio #337: Build error on CYGWIN64

thinksomuch, 2025-09-15:
Hi,

Please refer to following compile errors.

```
user@DESKTOP-ACJTEK1:~/project/tio$ meson setup --wipe

ERROR: Must specify at least one directory name.
user@DESKTOP-ACJTEK1:~/project/tio$ meson compile -C build
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/user/project/tio/build
ninja: Entering directory `/home/user/project/tio/build'
[2/6] Compiling C object src/tio.exe.p/tty.c.o
FAILED: [code=1] src/tio.exe.p/tty.c.o
cc -Isrc/tio.exe.p -Isrc -I../src -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/lua5.3 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=gnu99 -O3 -pthread -Wshadow -Wno-unused-result -MD -MQ src/tio.exe.p/tty.c.o -MF src/tio.exe.p/tty.c.o.d -o src/tio.exe.p/tty.c.o -c ../src/tty.c
../src/tty.c: In function ‘tty_wait_for_device’:
../src/tty.c:2364:21: error: label ‘error’ used but not defined
 2364 |                     goto error;
      |                     ^~~~
[5/6] Compiling C object src/tio.exe.p/script.c.o
ninja: build stopped: subcommand failed.
```

Thanks,

thinksomuch, 2025-09-15:
```
user@DESKTOP-ACJTEK1:~/project/tio$ git log -1
commit 3af4c5591e0183ea9871654ea4d62254ac23226d (HEAD -> master, origin/master, origin/HEAD)
Author: aiotter <git@aiotter.com>
Date:   Thu Aug 7 23:40:39 2025 +0900

    Fix redundant output on macOS
```

123swk123, 2026-02-19:
fixed in this pull request #363, Also added arbitrary baud rate support for MSYS2