tio/tio #340: Add optional Ctrl-C/Ctrl-D/~. exit support

mbUSC, 2025-09-20:
Reopening issue #203 

I have been using `tio` for a while and it has been a good serial monitor. 
The only issue is the prefix key. Having to do that every time I need to exit is absolutely atrocious

hrumhurum, 2026-10-27:
A typical use case is hardware development: typically an engineer just needs a quick and dirty way to exit the serial port terminal without too much fuzz.

Adding a `tio` command-line option like `--exit-shortcut ^c,^d` would save us some time and muscles.

Note that such exit needs to be configurable: some people want `Ctrl+C` but not `Ctrl+D` as an exit shortcut. Some hardware utilizes Ctrl+D semantics as "inner shell exit".

yabu76, 2026-10-27:
Couldn't a tio-wrapper like this serve as a workaround? :-)

exitio.sh
``` sh
#!/bin/bash

# A wrapper that launches `tio` with modified TTY settings

# Setting
EXIT_CHAR="^C" # preferred control character for termination.

# Start tio
TTYSET=$(stty -g)

tio --script "os.execute('stty isig intr ${EXIT_CHAR} kill undef eof undef susp undef start undef stop undef ')" "$@"

stty $TTYSET

```